utat-ss/HERON-pay #49: Bring up harness tests

brunofalmeida, 2019-09-20:


LornaLan, 2019-10-08:
I am going to create a new branch for pay-ssm v4 bringup tests. Not sure if doing this with harness or just manual test formats, will decide soon. Basically the steps for an entire bringup procedure are:
enable 10V boost and 6V boost
ADC readings (thank you @Tahreeem, will merge the branch `adc_testing` after uploading test)
environmental sensor readings (this should be minimal change from before)

After the above are done, we can move on to something more complicated:
heaters control (@mlin331 is working on this)
motors control
interface with pay-optical
CAN communication

brunofalmeida, 2019-10-09:
If you're ever unsure about how to test something I recommend just going with a manual test, then you might choose to migrate it to a harness test later if it makes sense.

closed 2020-01-14 by LornaLan